PowerShell/PowerShell-IoT #26: Support other devices

tylerl0706, 2018-04-13:
Allow this to be the issue where folks as for specific devices 😄 

I'll start.

One of the Particle boards: https://www.particle.io/

NOTE: this is bounded by .NET Core's ability

Ayanmullick, 2020-05-30:
Support for Wilderness Labs' Medow would be nice too.

http://developer.wildernesslabs.co/Meadow/Meadow.Foundation/Getting_Started/

Ayanmullick, 2020-07-17:
A [PowerShell SDK for the Azure Sphere devices](https://feedback.azure.com/forums/915433-azure-sphere/suggestions/40895797-powershell-sdk-for-azsphere) would be nice too.

https://azure.microsoft.com/en-us/services/azure-sphere/get-started/

<details><summary>Seeed MT3620 Mini Dev Board</summary>

[image]

</details>