PowerShell/PowerShell-IoT #28: Get-GpioPin force the pin to 'low' level

jnury, 2018-04-15:
Hi,

Using Get-GpioPin on a pin will force its level to 'low'.

To reproduce (even if the pin is not connected to anything):

`Set-GpioPin -Id 0 -Value High`

The pin is high.

`Get-GpioPin -Id 0`

Will return:

`Id Value PinInfo
-- ----- -------
 0   Low Unosquare.RaspberryIO.Gpio.GpioPin`

And the pin is low :-/

tylerl0706, 2018-04-15:
This is going to sound crazy... but you cannot set a pin and then get it's value. GPIO doesn't really work this way. A pin is either a setter or a getter. 

Once you set a pin, it's marked as an "input" pin with the value of "high" (or "low").

Once you get a pin, it's marked as an "output" pin with no guarantee that it will hold the value of the previous set operation.

@anmenaga can explain this further.

jnury, 2018-04-15:
Hum, my bad, Arduino (bad) habits :-D

Maybe we should have an object 'GPIOPin' with a 'type' = input|output and a Value that we can get/set if the type is Output or simply get if the type is Input.

BTW: thank you for you fast and detailed answer ;-)

tylerl0706, 2018-04-15:
That's a great idea :) lets change the name of this issue to support that.

jnury, 2018-04-17:
After digging a bit in docs (I'm not quite familiar with Arduino GPIO), it seems that is really easy to read status of an output pin.

Example with GPIO 2:

```
echo "27" > /sys/class/gpio/export
cat /sys/class/gpio/gpio7/value
```

I'll have a look if it's possible to use existing WirePi/RaspberryIO method or PR something to these projects to support that ;-)

tylerl0706, 2018-04-17:
That would be awesome! Pull requests always welcome 😃