PowerShell/PowerShell-IoT #29: Get-I2CRegister or Set-I2CRegister crash PowerShell if device is unreachable

jnury, 2018-04-15:
Hi,

If the device provided to Get or Set-I2CRegister commands is not reachable, PowerShell crash with message:

```
*** Error in `pwsh': free(): invalid pointer: 0x76d35694 ***
Aborted
```

To reproduce (do not connect anything on I2C bus):

```
$Device = Get-I2CDevice -ID 0x54 -FriendlyName 'accelerometer'
Get-I2CRegister -Device $Device -Register 0x00
```

PowerShell version:

```
Name                           Value
----                           -----
PSVersion                      6.0.2
PSEdition                      Core
GitCommitId                    v6.0.2
OS                             Linux 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018
Platform                       Unix
PSCompatibleVersions           {1.0, 2.0, 3.0, 4.0...}
PSRemotingProtocolVersion      2.3
SerializationVersion           1.1.0.1
WSManStackVersion              3.0
```

tylerl0706, 2018-04-15:
Yep, this looks like a PowerShell bug. PowerShell should never crash. Can you supply the exception that is spit out? Or is the first message all you get?

jnury, 2018-04-15:
The first message is all I get.

tylerl0706, 2018-04-15:
Interesting. Ok! Let me see if I can get a little bit more info during a repro and then we'll move this over to PowerShell/PowerShell as I doubt we'll be able to catch the exception.

Once we're able to do so, we can handle a `NoRegisterFoundException`

tylerl0706, 2018-04-16:
This could also be aproblem with .NET Core itself.

jnury, 2018-04-17:
I encountered crashes even with a working address/register, it seems unpredictable and after it occurred, I have to restart the whole device...

It seems to occur less with PowerShell Core 6.1.0 preview than it does in 6.0.2. Definitively a strange case ...

ThmsRynr, 2018-04-20:
I can apparently replicate this (or at least a very similar) issue on PowerShell 6.1.0-preview.1, and version 0.1.1 of the Microsoft.PowerShell.IoT module. The same error described above by @jnury  happens if I try to get the register, but I get a different error just getting the device.

```PowerShell
Import-Module Microsoft.PowerShell.IoT
Get-I2CDevice -Id 0x46 #where 0x46 is the correct ID of my device
```
pwsh will crash and throw the error `Unable to select I2C device: Device or resource busy`.

Interestingly, if I do a `Get-I2CDevice` for an ID that is just made up, it will simply return an I2CDevice object with whatever FriendlyName and Id values you pass it.

Also, you can crash all the way out of pwsh another way that I found out by accident while verifying the last point.
```PowerShell
Import-Module Microsoft.PowerShell.IoT
Get-I2CDevice -Id 911 #Or some other value that isn't valid
```
pwsh will crash, throwing the error `Unable to select I2C device: Invalid argument`.

tylerl0706, 2018-04-20:
Thanks for the info @ThmsRynr!

anmenaga, 2018-04-27:
Original PS crash with `free(): invalid pointer` error is coming from [Unosquare.RaspberryIO.dll](https://github.com/unosquare/raspberryio/blob/master/src/Unosquare.RaspberryIO/Native/HardwareException.cs)
The line 26 `ExtendedMessage = Standard.StrError(errorCode);` is calling 'strerror' function from 'libc' library passing 0 (that it gets from `Marshal.GetLastWin32Error()`) as an argument.
This should be fixed in Unosquare.RaspberryIO code (I've tried passing `-1` as argument as well as commenting out the problem line - both fixed the crash). I'm not sure how we can workaround this at our layer.

DanielSSilva, 2018-05-07:
as stated on #37 , [this](https://gist.github.com/DanielSSilva/ea2d5ddbaffa1ff5576622ea4543badd) code is causing that very same error, although I can't find a pattern for that (aka, the exception is random)