PowerShell/PowerShell-IoT #31: Allow the registering of change events for GPIO pins

tylerl0706, 2018-04-16:
It would look something like this:

```powershell
Register-GpioPinChangeEvent -Type LowToHigh -ScriptBlock {
    Write-Host "do the thing"
}

# and

Register-GpioPinChangeEvent -Type HighToLow -ScriptPath foo.ps1
```

Under the hood it could use `Register-ObjectEvent` to allow the script execution to be in the same runspace.

Unosquare docs here:
https://github.com/unosquare/raspberryio#interrupts-and-callbacks

jnury, 2018-04-18:
Really useful to handle I2C devices that use interrupts pins !

matswi, 2020-06-16:
Would this make it possible to detect button being pressed?

TylerLeonhardt, 2020-06-16:
That's the idea, yes

matswi, 2020-06-16:
> That's the idea, yes

That would be really nice! :) I'm afraid I don't have the skills to make it but would be happy to help with testing it.
Is there any plan on moving forward with it?

DanielSSilva, 2020-06-16:
@matswi we are currently migrating to .NET IoT library.
The closest that we will have so far (_I think_) "out of the box" will be the possibility to register an event for changes
https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/GpioController.cs#L281-L296
There's no ETA I'm afraid