PowerShell/PowerShell-IoT #35: Set-i2cdevice overload to support arrays

DanielSSilva, 2018-04-21:
I think it would be good to have an overload that would accept an array of registers and an array of values to be set on those registers, 'freeing' the developer to have the necessity to create a loop to iterate each array and set each value

tylerl0706, 2018-04-21:
Thanks for this Daniel! Did you mean `Set-I2CRegister`?

DanielSSilva, 2018-04-21:
Oh yes... I wrote it from my memory 😅

tylerl0706, 2018-04-21:
This is the case of really bad documentation but this is already supported!

https://github.com/PowerShell/PowerShell-IoT/blob/master/src/Microsoft.PowerShell.IoT/Microsoft.PowerShell.IoT.cs#L185-L189

Both `Register` and `Data` are `ValueFromPipelineByPropertyName` 😄

tylerl0706, 2018-04-21:
So if you have a custom object array and pass that into Set-I2CRegister it should work:
```
$device = Get-I2CDevice -Id 0x44 -FriendlyName Foo

$arr = @(
[pscustomobject]@{
    Register = 0x12
    Data = @(1,2,3)
    Device = $device
},
[pscustomobject]@{
    Register = 0x77
    Data = @(2,3,4)
    Device = $device
})

$arr | Set-I2CRegister
```

Note: haven't tested that code ^ but I'm pretty sure it works.

DanielSSilva, 2018-04-22:
Oh I see... although it's only valid to set an array of values into a register, not a 1 to 1 relation, aka, 1 register 1 value. But this might only be useful in my case, i don't know...

DanielSSilva, 2018-04-23:
No. I want an array of 5 registers and an array of 5 values and use register[i] value[i]. But then again, I don't have much experience to say that this would be a common situation on I2C protocol... Might only be useful in my case