PowerShell/PowerShell-IoT #40: Detect if I2C is enable and possibly enable it if user agrees

DanielSSilva, 2018-05-09:
For what I've seen, enable the I2C "manually" _seems_ quite easy, only requiring to edit:
-  the `/etc/modules` file to add `i2c-dev` 
- the `/boot/config.txt` file and uncomment the `dtparam=i2c_arm=on` entry.

If this is all that it takes, when the user imports this module, we can check if the i2c is already enabled and display a warning if it's not. Even better would be to prompt the user if he wants to enable it (avoiding the process of running `sudo raspi-config` and search for that).

Does this seems feasible/ make sense?

anmenaga, 2018-05-09:
It would certainly be beneficial to have a PowerShell way of enabling I2C (and, as extension, other interfaces).
Though doing this during module import may not be the best place because user may be wanting to use interface other than I2C.
Currently with disabled I2C `Get-I2CDevice` will give `Unable to open I2C device: No such file or directory` error. At a minimum we need to improve this error message to something like "Unable to open I2C device; please enable I2C interface using ..."
Ideally, I think, we should implement 2 cmdlets like:
`Enable/Disable-ArmInterface -I2C/-SPI/<others>`
and maybe add a call to them when Get-I2CDevice gets above-mentioned error.

tylerl0706, 2018-05-09:
We should also make sure these aren't pi specific since we do eventually want to support other devices.

tylerl0706, 2018-05-10:
We can start with Pi specific commands :)