PowerShell/PowerShell-IoT #42: [Opinion request] Bluetooth Support

DanielSSilva, 2018-08-01:
Hey. As the title implies, I would like to get some opinions about supporting bluetooth on this module. As far as I've seen, there are some C# implementations on bluetooth serialport communication.
Disclaimer: I'm not really familiar with bluetooth protocols and whatsoever, so I don't really know the implications or even if it's doable (but at least it seems).

Context: There are some devices that use bluetooth protocol to communicate. In this case I was thinking about the Sphero droids that currently use JavaScript and/or Scratch. In this very same case, it's more complicated because they do not expose an API to communicate with the droids (but that's another question). [I've seen a implementation](https://github.com/igbopie/spherov2.js) to communicate with them that used a javascript library . Maybe we can use some of the C# bluetooth projects?

Does this make any sense?

tylerl0706, 2018-08-03:
I've thought about this some... I think what makes sense (if it's possible) is a general purpose Bluetooth module that could be used on any platform.

This would probably be separate from the PowerShell IoT module found in this repo. At least that's my thought.

Have you seen any .NET based Bluetooth libraries?

DanielSSilva, 2018-08-03:
https://github.com/jcurl/serialportstream
I've seen this one, although I must confess that I haven't read everything and see if it fits the needs

tylerl0706, 2018-08-03:
This is kinda outside my element but... Can that interact with the on-board bluetooth device via a serial port?

I don't see anything talking about Bluetooth.

@anmenaga do you have any experience here?

DanielSSilva, 2018-08-03:
Possibly a dumb question but would it be possible to use the serial port if it was a Bluetooth dongle? (USB)

anmenaga, 2018-08-03:
I have one project which is running BLE server (listening for broadcasts) on RPi3. And another one that uses classic Bluetooth for streaming data between 2 devices (RPiZero with USB Bluetooth dongle as sender; Android phone as reciever).
At the time when I researched this, the most popular opinion was to use [BlueZ library](https://learn.adafruit.com/install-bluez-on-the-raspberry-pi/installation) and that's what I used. From my experience it is a rather good library.
So for those droids... I would first search if there is a C# lib for RPi's Bluetooth (not much hope for this);
Then second thing would be to write a native lib (that is using BlueZ) that implements low-level droid-specific Bluetooth operations while providing simple high-level functions to be called from .NET/PowerShell.

For external modules:
I'm sure there are I2C/SPI Bluetooth modules, but usually they provide low-level APIs, which means that handling them in managed/PS code will have a significant performance hit - which is often not acceptable for a task at hand.

ssimontis, 2019-11-04:
@TylerLeonhardt A lot of BLE devices allow for interaction with the device itself by exposing a serial port interface. I can post some examples when I get home and have all my data sheets lying around.

This project sounds really cool and I have some experience with embedded devices, so let me know how I can be of service!

DanielSSilva, 2019-11-06:
Hi @ssimontis. Thank you for your interest!
This project was a little bit in "stand-by" for me but I intend to come back to it really soon!
I would say that we could search for a C# library that already does this and contribute to it if it does not do what we want.
On the other hand, we are planning to migrate this project to use [dotnet iot](https://github.com/dotnet/iot), so it might be worth to check with them if they have anything regarding Bluetooth!