PowerShell/PowerShell-IoT #48: impact of 'wiringPI - deprecated' as this library is build on top if it?

mi-hol, 2020-03-18:
[Deprecate annoncement](http://wiringpi.com/wiringpi-deprecated/)
> I will make a final release of wiringPi available soon – with the sources, but that’s that. No more public releases. I’ll still be maintaining it for my own uses and clients, but for everyone else, please look at for alternative GPIO library for on-going projects.
> 
> -Gordon, August, 2019
Is an alternative GPIO library considered to replace wiringPI ?

TylerLeonhardt, 2020-03-18:
Yep, we plan to move to https://github.com/dotnet/iot but haven't gotten around to it. @DanielSSilva was starting it/going to start it but I'm not sure how far he got.

DanielSSilva, 2020-03-18:
Hi!
Yes I've started to port our current cmdlets to use a new library. We will be using the dotnet IoT repo for that.
I've successfully implemented the I2C using this library, without braking anything.
As for the gpio, there will be some breaking changes (I expect to create an issue and documentation for that). I've started to implement that but there are some issues that are being addressed with the dotnet IoT team.
You can see the progress on the branch forked from the master (I'm on mobile and can't link it)

PR #51 (2020-04-10): Updates for v2 of the module
Main changes in this:
1) Removed dependency on 3rd party libraries in favor of using .NET Core IoT Libraries.
2) Updated GPIO and SPI cmdlets with new APIs:
* because of specifics of new .NET APIs and to keep cmdlets simple - added new GPIO cmdlet `Clear-GpioResources`; it is needed in very few situations; majority of scenarios do not need to use it.
* in SPI cmdlets removed `Channel` and added `BusId` and `ChipSelectLine` parameters (because of specifics of new .NET APIs).
3) To align with PowerShell 7.0 release:
* updated `TargetFramework` to `netcoreapp3.1`
* up [... cut]
closed 2020-04-10 by anmenaga