PowerShell/PowerShell-IoT #49: Breaking changes

DanielSSilva, 2020-03-30:
With the migration to the [dotnet/iot](https://github.com/dotnet/iot) framework, breaking changes will be introduced.
This issue aims to list those changes, why they were introduced, as well as what's the new behavior.
This is just to keep track and discussed such breaking changes. Those will be listed on the changelog when we release a new version to the gallery.

TylerLeonhardt, 2020-03-30:
I think this is a necessary breaking change given the circumstance we just need to make sure we update our docs accordingly.

DanielSSilva, 2020-03-30:
Before changing the text of this issue I've wrote this, and forgot to leave it here.

> This is just to keep track and discussed such breaking changes. Those will be listed on the changelog when we release a new version to the gallery.
In this case, both changelog and our docs

PR #51 (2020-04-10): Updates for v2 of the module
Main changes in this:
1) Removed dependency on 3rd party libraries in favor of using .NET Core IoT Libraries.
2) Updated GPIO and SPI cmdlets with new APIs:
* because of specifics of new .NET APIs and to keep cmdlets simple - added new GPIO cmdlet `Clear-GpioResources`; it is needed in very few situations; majority of scenarios do not need to use it.
* in SPI cmdlets removed `Channel` and added `BusId` and `ChipSelectLine` parameters (because of specifics of new .NET APIs).
3) To align with PowerShell 7.0 release:
* updated `TargetFramework` to `netcoreapp3.1`
* up [... cut]
closed 2020-04-10 by anmenaga