PowerShell/PowerShell-IoT #50: GpioPin changes

DanielSSilva, 2020-03-30:
On the new library, the GPIO is controlled by a `GpioController` instance ([here's the class](https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/GpioController.cs#L22)).
As far as I've understood, there should be only 1 controller "active" (1 instance at the time). For that reason, I've decided that it was a better approach to have the *-GpioPin to receive the controller, similar to what's being done with the I2C device. This avoids the overload of creating a controller per command execution, and allows us to keep context of the pins through the controller.
This also requires that we have a new cmdlet to create a Controller (`Get-GpioController` ?)

Another thing to keep in mind is that we should have a way do properly dispose the controller, if the user doesn't do so (see [here](https://github.com/dotnet/iot/blob/master/src/devices/Hcsr04/Hcsr04.cs#L148-L155)).

Another breaking change is that this library does not use the `PullMode` like we currently have. Instead, they have a `PinMode` ([here's the source](https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/PinMode.cs)). This is used when opening a pin/setting its mode
``` csharp
_controller.OpenPin(_echo, PinMode.Input);
_controller.OpenPin(_trigger, PinMode.Output);
```
([source](https://github.com/dotnet/iot/blob/master/src/devices/Hcsr04/Hcsr04.cs#L41-L42))

As a final note, we now have to pay special attention to opening/closing pins while developing our cmdlets, in order to avoid potential bugs where pins are left open, or are wrongly closed :)

Here's a snippet with an example on how to get a pin info
```powershell
$controller = Get-GpioController #if no scheme is specified, use the default one

$pin = Get-GpioPin -Controller $controller -Id 5 -PinMode Input
```

TylerLeonhardt, 2020-03-31:
I think we can simply hold a singleton/static field inside of the module.

inside of `Get-GpioPin`:

```csharp
if(s_gpioController == null)
{
    s_gpioController = new GpioController(...);
}

// do logic
```

DanielSSilva, 2020-03-31:
Assuming that we can do that (the static field) inside the module without any problem, I can agree with that. This will have to work for any GpioPin cmdlet, that being get, set or anything else we might have later ( like this `RegisterCallbackForPinValueChangedEvent` which I think that can be a game changer 😃 )

PR #51 (2020-04-10): Updates for v2 of the module
Main changes in this:
1) Removed dependency on 3rd party libraries in favor of using .NET Core IoT Libraries.
2) Updated GPIO and SPI cmdlets with new APIs:
* because of specifics of new .NET APIs and to keep cmdlets simple - added new GPIO cmdlet `Clear-GpioResources`; it is needed in very few situations; majority of scenarios do not need to use it.
* in SPI cmdlets removed `Channel` and added `BusId` and `ChipSelectLine` parameters (because of specifics of new .NET APIs).
3) To align with PowerShell 7.0 release:
* updated `TargetFramework` to `netcoreapp3.1`
* up [... cut]
closed 2020-04-10 by anmenaga