PowerShell/PowerShell-IoT #53: SPI implementation

DanielSSilva, 2020-04-13:
I'm opening this issue so that we can discuss the implementation of the SPI interface. I'm not familiar with this interface, but from my understanding (and from what I'm seeing on the dotnet repo), this seems that the SPI device could be created once and used many times, whereas with current implementation, each time we are sending that through SPI, we are creating a new device and then disposing it.
Would it make sense to have a similar approach to what we have on I2C ? Something like 
``` powershell
Get- SPIDevice -BusId 0 -Frequency 2400 ...
```

DanielSSilva, 2020-04-13:
Not only that, but it seems that [there are some bindings](https://github.com/dotnet/iot/blob/master/src/devices/Ws28xx/Ws28xx.cs#L29) (which we might or might not end up using) that require the SPI device as a parameter

anmenaga, 2020-04-15:
It is probably a good idea to make cmdlet interface for SPI similar to what we have for I2C.