PowerShell/PowerShell-IoT #63: Support Raspberry Pi OS / Raspbian Buster

franklesniak, 2020-09-24:
The title says it all :) These newer versions of Raspberry Pi OS (formerly known as Raspbian) have been out for over a year at this point; they should be supported.

DanielSSilva, 2020-09-24:
With the new version of PowerShell IoT, it's expected that this is already happening. But I'll keep this in mind when testing. I'm currently working on Ubuntu 64bit version, but it makes total sense to have a raspian version alongside

franklesniak, 2020-09-25:
Cool, thanks! It may be that it works fine on Raspberry Pi OS and the Readme just needs to be updated :) But I guess it depends on how much scrutiny is needed before it's declared "supported".

JetzeMellema, 2020-11-13:
It's worth to note that PowerShell Core doesn't support the current Raspbian version Buster. Support is currently limited to Stretch: https://docs.microsoft.com/en-us/powershell/scripting/install/installing-powershell-core-on-linux?view=powershell-7.1#raspbian

There's also an issue with WiringPi and the Pi 4 hardware: #64

anmenaga, 2021-02-03:
That support list for Raspbian is out of date. Recent PS release should work on Buster fine.

philiprusinov, 2021-02-23:
> That support list for Raspbian is out of date. Recent PS release should work on Buster fine.

Is there an updated list somewhere?