PowerShell/PowerShell-IoT #64: Using old version of WiringPi breaks Pi4 compatibility

JetzeMellema, 2020-11-13:
The module installs fine on Raspberry Pi 4, but it's not actually possible to interact with GPIO. Error message after running Get-GpioPin:

> Unable to determine hardware version. I see: Hardware   : BCM2711
> ,
>  - expecting BCM2708, BCM2709 or BCM2835.
> If this is a genuine Raspberry Pi then please report this
> to projects@drogon.net. If this is not a Raspberry Pi then you
> are on your own as wiringPi is designed to support the
> Raspberry Pi ONLY.
> 

This is caused by an older version of WiringPi that doesn't recognize the updated hardware in the Raspberry Pi 4. The author published a newer version (2.52) that adds support for the Pi 4: http://wiringpi.com/wiringpi-updated-to-2-52-for-the-raspberry-pi-4b/

I installed this version on my Pi and can confirm with gpio -v:

> gpio version: 2.52
> Copyright (c) 2012-2018 Gordon Henderson
> This is free software with ABSOLUTELY NO WARRANTY.
> For details type: gpio -warranty
> 
> Raspberry Pi Details:
>   Type: Pi 4B, Revision: 01, Memory: 2048MB, Maker: Sony
>   * Device tree is enabled.
>   *--> Raspberry Pi 4 Model B Rev 1.1
>   * This Raspberry Pi supports user-level GPIO access.

This does not resolve the issue, the cmdlets still fail. It would be great if this module could support the latest WiringPi version.

anmenaga, 2021-02-03:
An updated version of the module is in the repo (it uses .NET APIs instead of WiringPi), but not yet on PS Gallery.
For now, just clone the repo onto Pi4 and it should work:
`git clone https://github.com/PowerShell/PowerShell-IoT.git`
and then in PS import module with 
`Import-Module ./src/Microsoft.PowerShell.IoT/Microsoft.PowerShell.IoT.psd1`

philiprusinov, 2021-02-23:
> https://github.com/PowerShell/PowerShell-IoT.git

Hello,
At the link above only Raspberry Pi 2 and 3 are mentioned as supported. 
I've try to Import the updated version of module, it is on different directory, than your example and failed with the following error:

```PS
PS /home/pi/powershell/IoT> Import-Module ./PowerShell-IoT/src/Microsoft.PowerShell.IoT/Microsoft.PowerShell.IoT.psd1
Import-Module: The module to process 'Microsoft.PowerShell.IoT.dll', listed in field 'NestedModules' of module manifest '/home/pi/powershell/IoT/PowerShell-IoT/src/Microsoft.PowerShell.IoT/Microsoft.PowerShell.IoT.psd1' was not processed because no valid module was found in any module directory.
```

Next, I've build the module libraries with build.ps1 and get version 0.2.0 running, but it seems again, not supported on BCM2711 / Raspberry PI 4 

```PS
PS /home/pi/powershell/IoT/PowerShell-IoT/out/Microsoft.PowerShell.IoT> Get-GpioPin -Id 16
Get-GpioPin: Operation is not supported on this platform.
```